algyan/XIAOGYAN #6: ES基板のIssue

matsujirushi, 2022-12-24:
- [ ] U2のパッド幅が小さい？

matsujirushi, 2022-12-25:
### LEDマトリックス

**SCLK:**

[image]

**GRID1:**

[image]
[image]

tatsu1225, 2022-12-27:
ES1基板を頂きましたので，とりま手持ちの部品で可能な所まで組み立てました．
・VRの抵抗値はいくつがベターですか？-VINを0～1.15Vで変化させたいと思われましたので，とりあえず10K～100KでOKですかね？
・チップ抵抗・コンデンサの大きさは，0603/0804だと初学者には厳しそうｗ が，あえて小さいパーツで楽しんでもらうという趣向であればOKだと思いました 

[image]

matsujirushi, 2022-12-28:
@tatsu1225 

> VRの抵抗値はいくつがベターですか？

記入漏れていました。ご指摘ありがとうございます。手元は**VR1=10K**にしています。（ https://github.com/algyan/algyan_8th/issues/6#issue-1510063566 に追加しました）

> チップ抵抗・コンデンサの大きさは，0603/0804だと初学者には厳しそうｗ

入手性考えると、リード部品かチップ部品1608(mm)と思いますが、手元在庫が1608なのでw
Discussionページ立てました。
https://github.com/algyan/algyan_8th/discussions/10

tatsu1225, 2022-12-28:
あー、eagleだとinch表記で表示されるんで0603(1608(mm))すね。ふだん、チップ抵抗使わないので勘違いしとりました🙇(お恥ずかしい、pad見ればわかるやん😇

matsujirushi, 2022-12-28:
@tatsu1225 
あるあるですねw（インチ<->ミリ）

あと、LEDマトリックスは2種類を動かしてみた感触では、2色のほうが良さそうなのでそちらを採用しようと思っています。（フットプリントはコスト上がらないので両方用意）

1. そちらで使っているボタンの通販コード、教えてもらえませんか？（秋月じゃない？）
2. XIAO SAMD21ってことは、もしかしてスピーカー辺りを試してくれようとしている？😍

tatsu1225, 2022-12-28:
LEDマトリックスは2色ありだと思います。弊学でも(別の先生が)実習で使っている様子を見てますが、表現の幅が広がる印象持ってます。

タクトスイッチは秋月ですが10個パックです。
通販コードは P-02561

XIAO ESP32の手持ちが尽きてしまったので、ひとまず無印XIAOを刺してみただけ。なんて言えない😇
(スピーカーは同じものが手元にない&秋月も年末年始モードなので、 P-12494 でひとまず試してみようかと。

matsujirushi, 2022-12-28:
### U1.D7はPULLDOWNできるが、U1.D9はPULLDOWNできない

* U1.D7 ... ESP32C3のPin27(U0RXD)
* U1.D9 ... ESP32C3のPin15(GPIO9) - BOOTボタンにも接続、BOOTボタンが10Kでプルアップされている。

matsujirushi, 2022-12-30:
### LED1をアクティブLOWに変更

1. LED1のアノードとカソードを反対にする（R12側がアノード）
2. R12を左パッドだけに付ける
3. R12の空き側を3V3に接続

[image]

matsujirushi, 2022-12-30:
### パワーアンプENABLEの効果

DISABLEにすると、2.78mA -> 167uAに下がる。（つまり、効果あり）

[image]

matsujirushi, 2022-12-31:
### Grove I2Cの波形

ちょっと立ち上がりが遅いのが気になるけど、Groveモジュール側の話なので、、、問題ないと判断。

[image]

tatsu1225, 2023-01-29:
・JP4のピンヘッダは実装＆ショートジャンパ必須
→ 3.3Vが各回路へ供給されない．どうも電流計測用のジャンパとのこと
　 ES2では削除予定(と聞きました
・以下のサンプルの動作確認ができました:-)
　－ButtonAndLed
　－Encoder
　－LedMatrixTM1640
　－Speaker2

matsujirushi, 2023-02-02:
> JP4のピンヘッダは実装＆ショートジャンパ必須

PADにして、デフォルトをショートにしました。

[image]

closed 2023-03-19 by matsujirushi