algyan/XIAOGYAN #7: お手伝い募集

matsujirushi, 2022-12-24:


610t, 2022-12-24:
ハードウエアはあまり強くありません。
XIAOシリーズは好きで、よく使っています。
XIAO ESP32C3を使うのであれば、Scratchと一緒に遊べればいいなと思っています。
よろしくお願いします。

tatsu1225, 2022-12-25:
お、最近，ESP32 XIAOに手を出してるのもある＆matsujirushi さんの仕事っぷりを間近で眺めたいので手伝いまっす！
ハード，ファーム周りが生業ですが， matsujirushi さんと比べると浅めです
手伝いやすいところでいえば，手元に電子パーツ類や実装の為の環境はそこそこあるので，検証用の回路試作やら動作確認用ファームなど

motojinc25, 2022-12-26:
IoT Edge系があれば、どうですか？
よろしくお願いいたします。

atkogure, 2022-12-27:
企画＆設計の全体像・方向性に関する意見、電子部品の選定や入手性などなど、私でお役に立てそうな場面があれば、何なりとお声かけくださいませ。

matsujirushi, 2022-12-28:
https://github.com/algyan/algyan_8th/issues/7#issuecomment-1364596872
@610t さん、ご連絡ありがとうございます！
Scratchと連動いいですね！（わたし自身、やったことない．．．）
現時点は、機能面を見ていただいて、Discussionにご意見いただけると嬉しいです。（○○はいらないとか、○○がほしいとか）
ハード周りが落ち着いたら、サンプル基板を送ろうと思います。しばしお待ちを。

https://github.com/algyan/algyan_8th/issues/7#issuecomment-1364651277
@tatsu1225 さん、ご連絡ありがとうございます！
マニュアル作りとアプリケーションあたりで手伝ってもらいたいです。
動作確認も手伝っていただけるようなので、、、ES基板を送りました。気づいた点があれば https://github.com/algyan/algyan_8th/issues/6 にコメント願います。

https://github.com/algyan/algyan_8th/issues/7#issuecomment-1364863945
@motojinc25 さん、ご連絡ありがとうございます！
クラウド側は、、、まだイメージできていない状況です（汗）
IoT Edge絡めて、どんな使い方ができそうか考えてもらえないでしょうか？（可能なら、Discussionに投稿ください。）

https://github.com/algyan/algyan_8th/issues/7#issuecomment-1365691205
@atkogure さん、ご連絡ありがとうございます！
パーツセットや完成品の販売を確立してほしいです！！！
（Seeedでやれよと思うところはありますが、なかなかに難しいのでお手伝いいただきたく）

y-ohe, 2022-12-28:
得意なことを考えているうちに出遅れてしまいました。
一応、回路設計、基板設計から簡単なソフト実装や3Dプリンタでのケース設計まで一通り何でもこなせるつもりです(クラウドまわりはあまり強くないですが、なんとなくぐらいはできます)。
EAGLEの部品ライブラリ整備なんて所もお手伝いできます。
何かお手伝いすることがあれば、割り振って下さい。

matsujirushi, 2022-12-28:
https://github.com/algyan/algyan_8th/issues/7#issuecomment-1366556129
@y-ohe さん、ご連絡ありがとうございます！
早速で恐縮なのですが、、、Seeed Studio XIAO ESP32C3の、裏のパッド付きのEAGLEライブラリが欲しいです！！（スタンパーの信号は、表面実装とスルーホールの両方実装できる感じで）

TakSan0, 2023-01-01:
見た時、ウルトラビジー状態だったので、後で回答と思っててすっかり出遅れました。
組み込み系 SE なので、ハード周りのソフトなら手伝えますかね。
時期によっては忙しいかもですが…
ちなみに、ANGYAN５周年及び６周年基板で、はんだ付けもくもく会用のハードチェックソフトとか作った実績あります。

KentaMatsumoto-2021, 2023-01-03:
nRF52840 なら手伝っても良いよ。

matsujirushi, 2023-01-07:
https://github.com/algyan/algyan_8th/issues/7#issuecomment-1368525541
@TakSan0 さん、ご連絡ありがとうございます！
ES2基板が出来上がったタイミングに、ES2基板+Seeed Studio XIAO nRF52840の組み合わせで一通り動作するか確認してほしいです！（2月中旬頃の見込み）
出来そうでしょうか？

https://github.com/algyan/algyan_8th/issues/7#issuecomment-1369755449
@KentaMatsumoto-2021 さん、ご連絡ありがとうございます！
nRF52840はちょっと先になりますが、そのときに相談させていただきます。

TakSan0, 2023-01-09:
@matsujirushi さん、了解です。
２月中くらいまで別件で忙しく、微妙にそれが伸びる気もしますが、その後３月にかけては大丈夫かと思います。
XIAOシリーズは、AMD21、nRF52840、 ESP32C3 いずれも持っていますが、積んでいる（放置）状態になっているので、それまでに、ちょっといじってみようかなと思います。

matsujirushi, 2023-03-06:
ES3基板で、ようやく改変が収まりました。
https://github.com/algyan/XIAOGYAN/blob/main/hardware/algyan_8th.es3.pdf

@610t @tatsu1225 @y-ohe @TakSan0 
試してもらいたいので、ES3基板を郵送しようと思います。
matsujirushi@live.jp まで、送付先を送っていただけませんか？

matsujirushi, 2023-03-10:
@610t @tatsu1225 @y-ohe @TakSan0 
配布基板の製造着手を4/3に決めました。
https://github.com/algyan/XIAOGYAN#%E3%82%BF%E3%82%A4%E3%83%A0%E3%83%A9%E3%82%A4%E3%83%B3

そのため、**ES3基板のフィードバック受付期限が3/26**です。
よろしくお願いします。

TakSan0, 2023-03-15:
ハード大体組みあがったので試してみようかと意気込んでましたが、
githubにある ALGYAN 8周年基板用のデモソフト  [xiaogyan_examples](https://github.com/algyan/xiaogyan_examples) は XIAO ESP32C3 向けなのですね。アナログ出力のところで ledc とか ESP32 にしかないライブラリを使っていますし。

なので、XIAO nRF52840 で試すとそこでエラーとなり、そこをコメントアウトしても、他にライブラリ側で大量にエラーが出てコンパイル通りませんでした。
[image]
まぁ、コンパイルスイッチもないので ESP32C3 専用にしか整備できていないんだと思っていますが、
一つ一つ解決していく必要ありそうです。

長い闘いになりそうです・・・

とりま、実績のありそうな XIAO ESP32C3 でハードが問題無いかだけ動かしてみようかなと、
コンパイルかけてみるとなんか D10 が無いとか？のピン定義でそんなエラーを見たことないのでちょっとよくわかりませんが…

こちらも、はまりそうな予感です・・・
[image]

以上、初見の状況でした。いきなりの地雷でつまづきつつあります。

TakSan0, 2023-03-15:
> コンパイルかけてみるとなんか D10 が無いとか？のピン定義でそんなエラーを見たことないのでちょっとよくわかりませんが…

単に D10 -> 10 の様に数値にするだけでコンパイルは通りました。
とりあえず、MATRIX LED が付くのは確認できました。
なぜか音の出力とスイッチ入力がまだ確認できません。部品付け忘れか何かか、調べてみます。

matsujirushi, 2023-03-16:
@TakSan0 
お試しありがとうございます！

issueを2つ起票しました。
不足があれば、issueを起票願います。

matsujirushi, 2023-03-16:
@TakSan0 
以下のissueをご確認願います。
#32 #33