algyan/XIAOGYAN #32: デモソフトxiaogyan_examplesが対象にしているXIAOがどれか分からない

matsujirushi, 2023-03-16:
> [元記事](https://github.com/algyan/XIAOGYAN/issues/7#issuecomment-1470078660)

githubにある ALGYAN 8周年基板用のデモソフト [xiaogyan_examples](https://github.com/algyan/xiaogyan_examples) は XIAO ESP32C3 向けなのですね。アナログ出力のところで ledc とか ESP32 にしかないライブラリを使っていますし。

matsujirushi, 2023-03-16:
現状、ハードウェア、ソフトウェアとも、マイコンは**Seeed Studio XIAO ESP32C3**で動作確認しています。
他のXIAOによる確認は、

* Seeed Studio XIAO SAMD21 ... ESの頃に、アナログ出力を確認したのみ。ES3は未確認。
* Seeed Studio XIAO RP2040 ... 未確認。
* Seeed Studio XIAO nRF52840 ... 基板設計時に机上で考慮。実機では未確認。

です。

当面、わたしはSeeed Studio XIAO ESP32C3についてフォローしようと考えています。
他のXIAOも出来れば対応したいので、issueやpull requestをお願いします。

TakSan0, 2023-03-16:
了解しました。
ESP32C3版のサンプルプログラムがどのような動きをするかの確認が出来次第、他のシリーズへのポーティング試してみます。
まずは nRF52840 から。コンパイルエラー対処からボチボチと。

Seeed Studio XIAO RP2040 は持っていませんが、 SAMD21 はどこかにあったはずですね。探せば出てくるかも。余裕あれば。

TakSan0, 2023-03-16:
XIAO_nRF52840 で使う際の問題点２か所特定できました。

### 1. 線形代数ライブラリ [Eigen](http://eigen.tuxfamily.org/index.php?title=Main_Page) がXIAO nRF52840 に未対応
　　D0 ～ D3 を使用している箇所でコンパイルエラーが多数出ていました。
　　これらはピン定義なので XIAO nRF52840 では数値で定義しているのに、ライブラリ側は '&' をつけて数値の入った変数の
　　アドレスを参照する造りとなっていたからでした。
　　ライブラリの git 更新差分で似たようなエラーを修正した差分があったのでそれをまねることで治りました。
　　具体的には
`<Library Path>/ArduinoEigen/ArduinoEigenCommon.h`
の最後の方に、

```
#ifdef D0
#undef D0
static constexpr size_t D0 {0};
#endif
#ifdef D1
#undef D1
static constexpr size_t D1 {1};
#endif
#ifdef D2
#undef D2
static constexpr size_t D2 {2};
#endif
#ifdef D3
#undef D3
static constexpr size_t D3 {3};
#endif

```
を追加しました。
但し、ライブラリ内のことなので、皆がコンパイルできるようにする為には、
ライブラリ作者に pullリク 出すか、正式に対応依頼しないといけないと思います。　　

### 2. Speaker.cpp のサイン波出力部　
ESP32 系 では Tone() 関数が使えないのが理由で、ledcWriteTone() 関数を使って出力しているかと思いますが、
逆に ESP32 ではない場合は、Arduino標準の Tone() の方が使えるはずと思ってやってみたら、ちゃんと出力できました。
▽こんな感じで修正してみました。

```
void Speaker::begin()
{
    pinMode(speakerEnablePin_, OUTPUT);
    digitalWrite(speakerEnablePin_, HIGH);
#if defined(ARDUINO_XIAO_ESP32C3)
    ledcAttachPin(speakerPin_, ledControlChannel_);
#endif
}

void Speaker::setPowerSupply(bool on)
{
    digitalWrite(speakerEnablePin_, on ? LOW : HIGH);
}

void Speaker::setTone(int frequency)
{
#if defined(ARDUINO_XIAO_ESP32C3)
    ledcWriteTone(ledControlChannel_, frequency);
#elif defined(ARDUINO_Seeed_XIAO_nRF52840) || defined(ARDUINO_Seeed_XIAO_nRF52840_Sense)
    if (frequency) tone(speakerPin_, frequency);
    else noTone(speakerPin_);
#endif
}

```

一応両方できるようにコンパイルスイッチで分けてみてます。

本来 pullリク出さないといけないのですが、大昔にやったっきりしばらく pullリク出していないので、
ちょっとリハビリ必要な感じです。明日できたらやってみます。
特に 1. の方の pullリクは、ちょっとハードル高いですね ^^

[取り急ぎのご報告まで。[](url)](url)

matsujirushi, 2023-03-16:
@TakSan0 
ご確認ありがとうございます。
Seeed Studio XIAO nRF52840対応のDiscussionを起票して転記しました。 #34

TakSan0, 2023-03-17:
XIAO AMD21 ボードはまだ発掘できていないのですが、先ずはコンパイル通るかを確認してみました。
ボード定義 "Seeeduino XIAO" を選択して試しています。

こちらは、ピンの定義自体が無いらしく Xiaogyan 辺りで D0～D10 や A0 ～ A10 の使用箇所でエラーが出るのと、
Eigen ライブラリコンパイル時に、別のエラーが多数出ますね。

#35
でおっしゃっていた 「案3. ArduinoEigenの使用をやめる。」　も検討しないといけないかもですね。
個々にエラーの詳細は確認できていませんが、こんな感じです。

別の issue 立てるべきかなとは思いますが、取り急ぎの報告です。

```
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h: In function 'void Eigen::internal::sparselu_gemm(Eigen::Index, Eigen::Index, Eigen::Index, const Scalar*, Eigen::Index, const Scalar*, Eigen::Index, Scalar*, Eigen::Index)':
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:31:3: error: expected primary-expression before 'enum'
   enum {
   ^~~~
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:40:20: error: 'RK' was not declared in this scope
   Index d_end = (d/RK)*RK;    // number of columns of A (rows of B) suitable for full register blocking
                    ^~
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:41:20: error: 'RN' was not declared in this scope
   Index n_end = (n/RN)*RN;    // number of columns of B-C suitable for processing RN columns at once
                    ^~
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:41:20: note: suggested alternative: '_N'
   Index n_end = (n/RN)*RN;    // number of columns of B-C suitable for processing RN columns at once
                    ^~
                    _N
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:58:30: error: 'BM' was not declared in this scope
   for(Index ib=i0; ib<m; ib+=BM)
                              ^~
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:58:30: note: suggested alternative: 'B'
   for(Index ib=i0; ib<m; ib+=BM)
                              ^~
                              B
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:61:37: error: 'SM' was not declared in this scope
     Index actual_b_end1 = (actual_b/SM)*SM;                   // actual number of rows suitable for peeling
                                     ^~
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:61:37: note: suggested alternative: 'SS'
     Index actual_b_end1 = (actual_b/SM)*SM;                   // actual number of rows suitable for peeling
                                     ^~
                                     SS
In file included from d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/SparseLU:28:0,
                 from d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/Sparse:29,
                 from d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/Eigen:2,
                 from d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen.h:10,
                 from D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\LedMatrix.hpp:11,
                 from D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\LedMatrixAGFX.cpp:11:
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h: In function 'void Eigen::internal::sparselu_gemm(Eigen::Index, Eigen::Index, Eigen::Index, const Scalar*, Eigen::Index, const Scalar*, Eigen::Index, Scalar*, Eigen::Index)':
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:31:3: error: expected primary-expression before 'enum'
   enum {
   ^~~~
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:40:20: error: 'RK' was not declared in this scope
   Index d_end = (d/RK)*RK;    // number of columns of A (rows of B) suitable for full register blocking
                    ^~
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:41:20: error: 'RN' was not declared in this scope
   Index n_end = (n/RN)*RN;    // number of columns of B-C suitable for processing RN columns at once
                    ^~
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:41:20: note: suggested alternative: '_N'
   Index n_end = (n/RN)*RN;    // number of columns of B-C suitable for processing RN columns at once
                    ^~
                    _N
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:58:30: error: 'BM' was not declared in this scope
   for(Index ib=i0; ib<m; ib+=BM)
                              ^~
In file included from d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/SparseLU:28:0,
                 from d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/Sparse:29,
                 from d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/Eigen:2,
                 from d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen.h:10,
[... cut ...]
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\ALGYAN8_ES3_Test.ino:65:32: error: request for member 'read' in 'XiaogyanClass::buttonA', which is of non-class type 'int'
     buttonA = Xiaogyan.buttonA.read() == LOW;
                                ^~~~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\ALGYAN8_ES3_Test.ino:66:32: error: request for member 'read' in 'XiaogyanClass::buttonB', which is of non-class type 'int'
     buttonB = Xiaogyan.buttonB.read() == LOW;
                                ^~~~
Compilation error: 'D10' was not declared in this scope
```

matsujirushi, 2023-03-17:
> こちらは、ピンの定義自体が無いらしく Xiaogyan 辺りで D0～D10 や A0 ～ A10 の使用箇所でエラーが出るのと、
> Eigen ライブラリコンパイル時に、別のエラーが多数出ますね。

ご確認ありがとうございます。
このissueとは違うので、新たにissueを起票願います。（1つのissueに混ぜて書いてあると、対応漏れしてしまうため）

> 取り急ぎの報告です。

こちらはissueではないので、、、Discussionに起票願います。