algyan/XIAOGYAN #33: xiaogyan_examplesでerror: 'D10' was not declared in this scope

matsujirushi, 2023-03-16:
> [元記事](https://github.com/algyan/XIAOGYAN/issues/7#issuecomment-1470078660)

とりま、実績のありそうな XIAO ESP32C3 でハードが問題無いかだけ動かしてみようかなと、
コンパイルかけてみるとなんか D10 が無いとか？のピン定義でそんなエラーを見たことないのでちょっとよくわかりませんが…

[image]

matsujirushi, 2023-03-16:
ボードの選択が間違っているように見えます。
`XIAO_ESP32C3`を選んで、再度確認願います。

[image]

### 参考リンク

* [Getting Started with Seeed Studio XIAO ESP32C3 - Software setup](https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started#software-setup)

TakSan0, 2023-03-16:
確かにボード設定ミスっていますね。
ちゃんとソースを元の状態に戻して、正しいボード設定ではコンパイル通ることが確認できました。
すみません、"参考リンク"に記載いただいた手順、ちゃんと見るべきでした。
ありがとうございます。

おかげで ESP32C3 版では、どのような動きになるかは確認できました。
一通り、マトリックスの動作、スピーカーからのTone再生、ボタン2つ組み合わせで３種類の音程が出る事、ジョグダイヤルで点滅速度が変わること、ジョグダイヤル内の LED が光る事が全て確認できました。
ありがとうございます。