algyan/XIAOGYAN #35: GeneralBlockPanelKernel.h:1270:73: note: in expansion of macro 'D0'

matsujirushi, 2023-03-17:
> [元記事](https://github.com/algyan/XIAOGYAN/discussions/34)

ターゲットボードを`Seeed Studio XIAO nRF52840`でコンパイルすると、ArduinoEigenの部分でエラーが発生する。

```
C:\Users\XXXXXXXXXX\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\1.1.1\variants\Seeed_XIAO_nRF52840/variant.h:46:16: error: lvalue required as unary '&' operand
   46 | #define D0 (0ul)
      |                ^
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1270:73: note: in expansion of macro 'D0'
 1270 |           peeled_kc_onestep(1, blA, blB, traits, &A1, &rhs_panel, &T0, &D0, &D1, &D2, &D3);
      |                                                                         ^~
C:\Users\XXXXXXXXXX\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\1.1.1\variants\Seeed_XIAO_nRF52840/variant.h:47:16: error: lvalue required as unary '&' operand
   47 | #define D1 (1ul)
      |                ^
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1270:78: note: in expansion of macro 'D1'
 1270 |           peeled_kc_onestep(1, blA, blB, traits, &A1, &rhs_panel, &T0, &D0, &D1, &D2, &D3);
      |                                                                              ^~
C:\Users\XXXXXXXXXX\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\1.1.1\variants\Seeed_XIAO_nRF52840/variant.h:48:16: error: lvalue required as unary '&' operand
   48 | #define D2 (2ul)
      |                ^
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1270:83: note: in expansion of macro 'D2'
 1270 |           peeled_kc_onestep(1, blA, blB, traits, &A1, &rhs_panel, &T0, &D0, &D1, &D2, &D3);
      |                                                                                   ^~
C:\Users\XXXXXXXXXX\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\1.1.1\variants\Seeed_XIAO_nRF52840/variant.h:49:16: error: lvalue required as unary '&' operand
   49 | #define D3 (3ul)
      |                ^
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1270:88: note: in expansion of macro 'D3'
 1270 |           peeled_kc_onestep(1, blA, blB, traits, &A1, &rhs_panel, &T0, &D0, &D1, &D2, &D3);
      |                                                                                        ^~
C:\Users\XXXXXXXXXX\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\1.1.1\variants\Seeed_XIAO_nRF52840/variant.h:46:16: error: lvalue required as unary '&' operand
   46 | #define D0 (0ul)
      |                ^
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1272:73: note: in expansion of macro 'D0'
 1272 |           peeled_kc_onestep(3, blA, blB, traits, &A1, &rhs_panel, &T0, &D0, &D1, &D2, &D3);
      |                                                                         ^~
C:\Users\XXXXXXXXXX\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\1.1.1\variants\Seeed_XIAO_nRF52840/variant.h:47:16: error: lvalue required as unary '&' operand
   47 | #define D1 (1ul)
      |                ^
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1272:78: note: in expansion of macro 'D1'
 1272 |           peeled_kc_onestep(3, blA, blB, traits, &A1, &rhs_panel, &T0, &D0, &D1, &D2, &D3);
      |                                                                              ^~
C:\Users\XXXXXXXXXX\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\1.1.1\variants\Seeed_XIAO_nRF52840/variant.h:48:16: error: lvalue required as unary '&' operand
   48 | #define D2 (2ul)
      |                ^
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1272:83: note: in expansion of macro 'D2'
 1272 |           peeled_kc_onestep(3, blA, blB, traits, &A1, &rhs_panel, &T0, &D0, &D1, &D2, &D3);
      |                                                                                   ^~
C:\Users\XXXXXXXXXX\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\1.1.1\variants\Seeed_XIAO_nRF52840/variant.h:49:16: error: lvalue required as unary '&' operand
   49 | #define D3 (3ul)
      |                ^
d:\DEVELOP\XXXXXXXXX\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/Core/products/GeneralBlockPanelKernel.h:1272:88: note: in expansion of macro 'D3'
 1272 |           peeled_kc_onestep(3, blA, blB, traits, &A1, &rhs_panel, &T0, &D0, &D1, &D2, &D3);
      |                                                                                        ^~
```

matsujirushi, 2023-03-17:
**Adafruit_nRF52_Arduino**で`#define`している`D0`, `D1`, `D2`, `D3`が、**ArduinoEigen**の内部に影響している。

https://github.com/Seeed-Studio/Adafruit_nRF52_Arduino/blob/ccd02aa73be1a9a08fb3cbc88ca448a365e6729e/variants/Seeed_XIAO_nRF52840/variant.h#L46%E3%83%BCL49

https://github.com/hideakitai/ArduinoEigen/blob/f6447f87dfa120eb2bbc8c3d251ef916f014d730/ArduinoEigen/Eigen/src/Core/products/GeneralBlockPanelKernel.h#L1241

案1. Adafruit_nRF52_Arduinoの`#define`を`constexpr`に変更。→問題無さそうだがこのプラットフォーム利用者全員に影響。OwnerはSeeed-Studio。
案2. ArduinoEigenの変数名を変更。→[2022/11のissue](https://github.com/hideakitai/ArduinoEigen/issues/2)が対応されていない。Ownerはhideakitai。
案3. ArduinoEigenの使用をやめる。→OwnerがALGYAN、かつ、Eigenの利用箇所は限定的なので改修は楽。

matsujirushi, 2023-03-17:
案1をこちらに用意した。
https://github.com/algyan/Adafruit_nRF52_Arduino

matsujirushi, 2023-03-17:
マージされました。
もし、問題があればreopenしてください。

closed 2023-03-17 by matsujirushi

TakSan0, 2023-03-17:
マージされたとのことで早速アップデートして試してみようとして、ボードマネージャーを開いて検索すると、
"Seeed nRF52 Boards by Seeed Studio" の最新は 1.1.1 となっており、最新が入っていると認識されているので
更新できませんでした。
一旦、バージョンを1.1.0 に戻して再度1.1.1 を更新したら変わるかなと試してみたのですが、該当ソースを見ると
修正されたはずの箇所は、マクロのままでした。

もしかして、上記更新方法は間違っていますでしょうか？
公式のボードとしては正式にはアップデートされていないので、手動でコピーしないといけないとかありますか？

余り、ボードマネージャーの更新の仕組みが判っていないので、的外れなことを言っていたら申し訳ないのですが、
[platform.txt](https://github.com/Seeed-Studio/Adafruit_nRF52_Arduino/blob/master/platform.txt)
にある、
`version=1.1.1
`
のあたりは更新する必要ないのでしょうか？

matsujirushi, 2023-03-17:
@TakSan0 
まだリリースされていないので、ボードマネージャーからアップデートはできないです。
一旦、ボードマネージャーでインストールして、同一フォルダにGitHubから最新を取得して上書きすることで、GitHub上の最新を試すことができます。

```
cd C:\Users\takashi\AppData\Local\arduino15\packages\Seeeduino\hardware\nrf52
rmdir /S /Q 1.1.1
git clone https://github.com/Seeed-Studio/Adafruit_nRF52_Arduino 1.1.1
cd 1.1.1
git submodule init
git submodule update
```

TakSan0, 2023-03-18:
すみません。よくわかってなくて…

rmdir /S /Q 1.1.1
で、書き込み用のツールや設定やヘッダーを消してしまうからなのか
```
Alternatives for Adafruit_USBD_CDC.h: []
ResolveLibrary(Adafruit_USBD_CDC.h)
  -> candidates: []
In file included from D:\DEVELOP\TakSan\MicroComputer\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\ALGYAN8_ES3_Test.ino:10:
C:\Users\TakSan\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\1.1.1\cores\nRF5/Arduino.h:65:10: fatal error: Adafruit_USBD_CDC.h: No such file or directory
   65 | #include "Adafruit_USBD_CDC.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```
といったエラーになって書き込み前くらいの所でエラーで止まってしまいました。

あと、そのまま教えていただいたコマンド 
```
submodule init
submodule update 

```
を実行しただけでは、なぜかファイルが更新されなかったので、結局
元の 1.1.1 配下のファイルをすべて元に戻したあと
一旦 ブラウザから .zip をダウンロードして解凍後　該当箇所に上書きしました。

その状態で Eigen に何も手を入れていない状態でも、コンパイルエラーとならずに
書き込み成功して動作することが確認できました。

matsujirushi, 2023-03-19:
@TakSan0 
`fatal error: Adafruit_USBD_CDC.h: No such file or directory`の件、issue起票しました。 #39