algyan/XIAOGYAN #36: XIAOGYAN のライブラリを XIAO AMD21 向けでコンパイルするとピン定義な無くてエラーとなる。

TakSan0, 2023-03-18:
元記事 [#32](https://github.com/algyan/XIAOGYAN/issues/32#issuecomment-1473753129)

XIAOGYAN のライブラリを XIAO AMD21 向け ("Seeeduino XIAO") のボード設定でコンパイルすると、ピン定義関連のエラーが発生して下記の様なコンパイルエラーとなります。

```
In file included from D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.cpp:1:0:
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp: At global scope:
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:22:51: error: 'D10' was not declared in this scope
     static constexpr uint8_t TM1640_DIN_PIN     = D10;
                                                   ^~~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:22:51: note: suggested alternative: 'A10'
     static constexpr uint8_t TM1640_DIN_PIN     = D10;
                                                   ^~~
                                                   A10
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:23:51: error: 'D8' was not declared in this scope
     static constexpr uint8_t TM1640_SCLK_PIN    = D8;
                                                   ^~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:23:51: note: suggested alternative: 'A8'
     static constexpr uint8_t TM1640_SCLK_PIN    = D8;
                                                   ^~
                                                   A8
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:24:51: error: 'D0' was not declared in this scope
     static constexpr uint8_t SPEAKER_PIN        = D0;
                                                   ^~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:24:51: note: suggested alternative: 'B0'
     static constexpr uint8_t SPEAKER_PIN        = D0;
                                                   ^~
                                                   B0
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:25:51: error: 'D3' was not declared in this scope
     static constexpr uint8_t SPEAKER_ENABLE_PIN = D3;
                                                   ^~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:25:51: note: suggested alternative: 'A3'
     static constexpr uint8_t SPEAKER_ENABLE_PIN = D3;
                                                   ^~
                                                   A3
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:26:51: error: 'D1' was not declared in this scope
     static constexpr uint8_t ENCODER_A_PIN      = D1;
                                                   ^~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:26:51: note: suggested alternative: 'B1'
     static constexpr uint8_t ENCODER_A_PIN      = D1;
                                                   ^~
                                                   B1
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:27:51: error: 'D2' was not declared in this scope
     static constexpr uint8_t ENCODER_B_PIN      = D2;
                                                   ^~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:27:51: note: suggested alternative: 'A2'
     static constexpr uint8_t ENCODER_B_PIN      = D2;
                                                   ^~
                                                   A2
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:28:51: error: 'D6' was not declared in this scope
     static constexpr uint8_t LED_PIN            = D6;
                                                   ^~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:28:51: note: suggested alternative: 'A6'
     static constexpr uint8_t LED_PIN            = D6;
                                                   ^~
                                                   A6
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:29:51: error: 'D9' was not declared in this scope
     static constexpr uint8_t BUTTON_A_PIN       = D9;
                                                   ^~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:29:51: note: suggested alternative: 'A9'
     static constexpr uint8_t BUTTON_A_PIN       = D9;
                                                   ^~
                                                   A9
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:30:51: error: 'D7' was not declared in this scope
     static constexpr uint8_t BUTTON_B_PIN       = D7;
                                                   ^~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:30:51: note: suggested alternative: 'A7'
     static constexpr uint8_t BUTTON_B_PIN       = D7;
                                                   ^~
                                                   A7
[... 86 lines cut ...]
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:30:51: note: suggested alternative: 'A7'
     static constexpr uint8_t BUTTON_B_PIN       = D7;
                                                   ^~
                                                   A7
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:45:33: error: could not convert template argument 'XiaogyanClass::LED_PIN' from 'const uint8_t {aka const unsigned char}' to 'int'
     static GpioOutputPin<LED_PIN> led;
                                 ^
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:46:37: error: could not convert template argument 'XiaogyanClass::BUTTON_A_PIN' from 'const uint8_t {aka const unsigned char}' to 'int'
     static GpioInputPin<BUTTON_A_PIN> buttonA;
                                     ^
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\Xiaogyan.hpp:47:37: error: could not convert template argument 'XiaogyanClass::BUTTON_B_PIN' from 'const uint8_t {aka const unsigned char}' to 'int'
     static GpioInputPin<BUTTON_B_PIN> buttonB;
                                     ^
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\ALGYAN8_ES3_Test.ino: In function 'void loop()':
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\ALGYAN8_ES3_Test.ino:58:18: error: request for member 'write' in 'XiaogyanClass::led', which is of non-class type 'int'
     Xiaogyan.led.write(millis() % 1000 < 200 ? LOW : HIGH);
                  ^~~~~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\ALGYAN8_ES3_Test.ino:65:32: error: request for member 'read' in 'XiaogyanClass::buttonA', which is of non-class type 'int'
     buttonA = Xiaogyan.buttonA.read() == LOW;
                                ^~~~
D:\DEVELOP\XXXXXXXXX\Arduino\ESP32\ALGYAN_Board\ALGYAN8_ES3_Test\ALGYAN8_ES3_Test.ino:66:32: error: request for member 'read' in 'XiaogyanClass::buttonB', which is of non-class type 'int'
     buttonB = Xiaogyan.buttonB.read() == LOW;

```
対処方法はまだ未検討ですが、ひとまず issue 起票します。

matsujirushi, 2023-03-19:
### Note
#37

TakSan0, 2023-03-19:
ちょっとエラーの詳細みてみましたが解決できていません。

一番最初の、
```
d:\DEVELOP\XXXXXXXXXX\MicroComputer\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h: In function 'void Eigen::internal::sparselu_gemm(Eigen::Index, Eigen::Index, Eigen::Index, const Scalar*, Eigen::Index, const Scalar*, Eigen::Index, Scalar*, Eigen::Index)':
d:\DEVELOP\XXXXXXXXXX\MicroComputer\Arduino\libraries\ArduinoEigen-master/ArduinoEigen/Eigen/src/SparseLU/SparseLU_gemm_kernel.h:31:3: error: expected primary-expression before 'enum'
   enum {
   ^~~~

```
が、全ての始まりみたいですが、該当箇所見ると、
```
void sparselu_gemm(Index m, Index n, Index d, const Scalar* A, Index lda, const Scalar* B, Index ldb, Scalar* C, Index ldc)
{
  using namespace Eigen::internal;
  
  typedef typename packet_traits<Scalar>::type Packet;
  enum {
    NumberOfRegisters = EIGEN_ARCH_DEFAULT_NUMBER_OF_REGISTERS,
    PacketSize = packet_traits<Scalar>::size,
    PM = 8,                             // peeling in M
    RN = 2,                             // register blocking
    RK = NumberOfRegisters>=16 ? 4 : 2, // register blocking
    BM = 4096/sizeof(Scalar),           // number of rows of A-C per chunk
    SM = PM*PacketSize                  // step along M
  };
  Index d_end = (d/RK)*RK;    // number of columns of A (rows of B) suitable for full register blocking
  Index n_end = (n/RN)*RN;    // number of columns of B-C suitable for processing RN columns at once
  Index i0 = internal::first_default_aligned(A,m);
  
  eigen_internal_assert(((lda%PacketSize)==0) && ((ldc%PacketSize)==0) && (i0==internal::first_default_aligned(C,m)));

```
の様になっており、テンプレート使用箇所のどこかで何かが不足していて、enum を正常に定義できずに発生しているという気はします。ただ、テンプレートが 絡むC++のエラーは、エラー箇所と原因が特定しにくくてちょっと苦手です...
ちゃんと解決するには Eigen のライブラリ実装方法をもう少し理解しないといけなくて、本筋から乖離していくかもですね。

因みに、 Eigen を使っている理由はソースを見る限りでは、 MatrixLED の描画データを回転させるためですよね？
で今の所回転したい角度は 90°単位のようですが、それよりも細かく設定できるようにしたかったという事でしょうか？
そうではなく90°単位でいいのであれば、２重ループで回して直接スワップ処理で実現しても問題ない気がします。
（改修時間の費用対効果的にはベターな気がします…）

matsujirushi, 2023-03-21:
> テンプレート使用箇所のどこかで何かが不足していて

`error: expected primary-expression before 'enum'`なので、`typedef typename packet_traits<Scalar>::type Packet;`が文法的に通っていないのでは？想像ですが、C++のバージョンが古いとか。

> Eigen を使っている理由はソースを見る限りでは

はい、回転と反転に使っています。

> 改修時間の費用対効果的にはベターな気がします

いまはハードウェアに影響するissueが最優先なので、Eigen絡みはざくっとコメントにして進めるのが良いと思います。