algyan/XIAOGYAN #41: ES3->配布基板のissue

matsujirushi, 2023-03-19:
- [ ] GNDビアを追加
- [ ] 部品番号の振り直し->シルクの調整

tatsu1225, 2023-03-21:
部品表とES3基板を確認しました．10KΩの抵抗は1608でFIXですか？
関連： 1608チップ抵抗の入手方法 #23

tatsu1225, 2023-03-21:
既に確認済みかもしれませんが，U10 のEC12E(Panel)[P-06357]ですが，ノンクリックタイプで良ければ在庫ありなようです
https://akizukidenshi.com/catalog/g/gP-06358/

tatsu1225, 2023-03-21:
ソケットは部品表に記載なしですが，公式的には直付けで付けたい人はご自由に．ですかね？
組立手順書を作る際にどちらにしようかと思ってます．

【ソケットありにすると...？】
　①部品代が安くなる
　**②他のXIAOに差し替えて遊べる**
　③ユニバーサル基板に回路拡張した際にXIAO壊したら交換できる
　④オペアンプは逆に半田付けしてしまった時の保険になる？
　⑤高さがでる　→　ロータリエンコーダの背が高いので何とも...

まとめてて⑤があまり強い理由にならないので，②の強みを推してソケットありを標準にしても良いかも．と思いました
あとは①との兼ね合いですかね

matsujirushi, 2023-03-22:
> 部品表とES3基板を確認しました．10KΩの抵抗は1608でFIXですか？ 関連： 1608チップ抵抗の入手方法 #23

1608とします。
こちらにも記入しました。 #23

matsujirushi, 2023-03-22:
> 既に確認済みかもしれませんが，U10 のEC12E(Panel)[P-06357]ですが，ノンクリックタイプで良ければ在庫ありなようです https://akizukidenshi.com/catalog/g/gP-06358/

触ってみるとわかるのですが、ノンクリックは回転トルクが大きすぎるんです...。

tatsu1225, 2023-03-22:
> > 既に確認済みかもしれませんが，U10 のEC12E(Panel)[P-06357]ですが，ノンクリックタイプで良ければ在庫ありなようです https://akizukidenshi.com/catalog/g/gP-06358/
> 
> 触ってみるとわかるのですが、ノンクリックは回転トルクが大きすぎるんです...。

あー、確かに。手元に同型番の部品を持っていますが、確かにその通りですね。では、現行のU10を標準として考えます。

EC12E、基板の穴にもピッタリハマって収まりが良さそうなのに惜しい…🥲🥲🥲

y-ohe, 2023-03-26:
@matsujirushi さん
ES3基板のR7、R18は、10KΩ×2(=5KΩ)だとロータリーエンコーダのLEDが非常に暗いので、100Ω1個ぐらいにした方が良さそうです。
秋月電子通商に1608の100Ω抵抗はあるので、これを購入して使ってもらえば良いと思います。
↓
https://akizukidenshi.com/catalog/g/gR-11789/

matsujirushi, 2023-03-26:
@y-ohe さん、アドバイスありがとうございます。

[基板配布時に10Kを同封](https://github.com/algyan/XIAOGYAN/discussions/23#discussioncomment-5391956)するので、部品リストは10Kのままにしておき、改造ネタで100換装という扱いにしようと思います。 （業務連絡 > @tatsu1225 )

y-ohe, 2023-03-27:
LED1とR12もありますしね。
LED1には、手持ちの5mm赤LEDを実装して、ロータリーエンコーダーのLEDとの同時点灯を確認しました。

tatsu1225, 2023-03-27:
期日間に合わずスミマセン．遅くなりましたが，ES3の実装終わりました！以下の動作を確認済み．
  1. マトリクスLED
  2. ボタン(A,B)
  3. スピーカ
  4. エンコーダ
  5. Groveコネクタ(SHT31で動作を確認)

実装時に気になった確認事項は以下の通り．
  - JP1は3.3VをショートしましたがOKですか？
  - JP2はSPKEAKER-ENABLEをショートしましたがOKでしたか？
  - スピーカ鳴動時のボリュームですが，ツマミを真ん中付近にした時が最大音量．左右のどちらかに回しきる時が最小音量の認識でOKですか？

組立手順的な観点からの指摘事項は以下の点です．が，現状のままでも問題無い内容です．
 - 初学者向けに振るならば，JP1, JP2はデフォルトでショートさせても良いかも？ → 工程は増えるが組立手順で提示すれば問題なし
 - 0Ω抵抗はハンダのみでショートさせることは出来ないので，抵抗の足などを利用することが推奨 → 初学者向けには0Ω抵抗の購入がbetter
 - マトリクスLEDはシルクで向きが分かるようにしても良いかも？ → 組立手順の工夫で何とでもなる
 
[image]

tatsu1225, 2023-03-27:
しかし，U10の設計はお見事ですね!! クリッカブルなEC12E(Panel)が在庫切れなのが非常に惜しい...!!(組立手順には載せておこうと思っています
[image]
[image]

tatsu1225, 2023-03-27:
ES3基板の組立には写真を撮りながらでしたが，ザックリ3時間ぐらいでした．
手順書アリで組みなれた人なら2時間かからないぐらいかな？
どの程度の時間かかりました？ > @y-ohe @matsujirushi

y-ohe, 2023-03-27:
> 1. [ピンヘッダ　１×４０　（４０Ｐ）](https://akizukidenshi.com/catalog/g/gC-00167/)
> 2. [分割ロングピンソケット　１×４２　（４２Ｐ）](https://akizukidenshi.com/catalog/g/gC-05779/)

ここは、私は高さを押さえたくて、
1. [ロープロファイルピンヘッダ（低オス）　１×４０（４０Ｐ）　７．７ｍｍ](https://akizukidenshi.com/catalog/g/gC-02900/)
2. [シングルピンソケット　（低メス）　１×２０　（２０Ｐ）](https://akizukidenshi.com/catalog/g/gC-03138/)
を使っています。

> どの程度の時間かかりました？ > @y-ohe @matsujirushi

ちゃんと時間を計らなかったので、感覚的な数字になりますが、**1時間**ぐらいだったと思います。
やはり、マトリクスLEDは、組み付けの向きがそのままでは分からず、 @matsujirushi さんの組立て後の写真を見てサイドに部品のマーキングがなかったところから何とか間違わずにはんだ付けができました。

matsujirushi, 2023-03-27:
@tatsu1225 
ご確認ありがとうございます！

> * JP1は3.3VをショートしましたがOKですか？

OKです。（データシートによるとU2は3.3V NGですが、、、動いているので3.3Vにしましょう。issueが上がったら対処ということで。）

> * JP2はSPKEAKER-ENABLEをショートしましたがOKでしたか？

OKです。

> * スピーカ鳴動時のボリュームですが，ツマミを真ん中付近にした時が最大音量．左右のどちらかに回しきる時が最小音量の認識でOKですか？

設計意図とは違いますが、誰からも指摘が無い&代案も無いので、振る舞いのとおりです。
v2までに、誰かが変更するのに期待しています。

> * 初学者向けに振るならば，JP1, JP2はデフォルトでショートさせても良いかも？ → 工程は増えるが組立手順で提示すれば問題なし

どちらを選択しているかを使用者に認識してほしいのでデフォルトでショートにしていません。

> * 0Ω抵抗はハンダのみでショートさせることは出来ないので，抵抗の足などを利用することが推奨 → 初学者向けには0Ω抵抗の購入がbetter

あとから改造しやすいように、0Ω抵抗としました。

> * マトリクスLEDはシルクで向きが分かるようにしても良いかも？ → 組立手順の工夫で何とでもなる

U3に"label side"のシルクを追加しました。

matsujirushi, 2023-03-27:
@tatsu1225 

> ES3基板の組立には写真を撮りながらでしたが，ザックリ2.5時間ぐらいでした． 手順書アリで組みなれた人なら1時間かからないぐらいかな？ どの程度の時間かかりました？ > @y-ohe @matsujirushi

どうだろう？2時間くらいかな。（作業合間に実装を進めたので分からないです）

matsujirushi, 2023-03-27:
FYI
U3の"label side"のシルク。
[image]

matsujirushi, 2023-03-27:
@tatsu1225 ピンヘッダとピンソケット、どの部品にしますか？
https://github.com/algyan/XIAOGYAN/issues/41#issuecomment-1485762724
https://github.com/algyan/XIAOGYAN/issues/41#issuecomment-1485977479

わたしが気になる点としては、XIAO ESP32C3を使い回すことも想定するなら、ブレッドボードに差しやすい細ピンにしたほうが良いと思いました。

全てを満たすパーツは無いと思いますので、お任せします。

tatsu1225, 2023-04-03:
すんません、遅くなりました
@matsujirushi さん案の細ピンヘッダ＆細ピン用の分割ロングピンソケットでいきましょう！
https://akizukidenshi.com/catalog/g/gC-04398/
https://akizukidenshi.com/catalog/g/gC-10073/

610t, 2023-04-25:
> ES3基板の組立には写真を撮りながらでしたが，ザックリ2.5時間ぐらいでした． 手順書アリで組みなれた人なら1時間かからないぐらいかな？ どの程度の時間かかりました？ > @y-ohe @matsujirushi

今日作成しました。
表面実装は初めてだったのですが、1.5時間で作成完了でした。

closed 2023-05-28 by matsujirushi