algyan/XIAOGYAN #46: Arduinoで作ったプログラムで、コンパイル時にスピーカー関係のエラーが出る。

610t, 2023-04-27:
現在、XIAOGYANでScratch3(正確にはStretch3 https://stretch3.github.io/ のMicrobit More拡張機能)を使えるようにするXIAOGYANbitLess https://github.com/610t/XIAOGYANbitLess を作っています。

スピーカーのArduinoプログラムでの利用についてお聞きしたいのです。

PlatformIOでは、platformio.iniの中で、-D SPEAKER=1としてスピーカーの種類を渡しています。
Arduino用に、以下のようにして、プログラム中で強制的にSPEAKER 1を使うようにしています。
~~~
#if !defined(SPEAKER)
#define SPEAKER 1
#endif
#include <Xiaogyan.hpp>
~~~

実際にスピーカーを利用してる部分のコードはこんな感じです。
~~~
// Tone handler
void playToneTask(void *args) {
  uint16_t freq = 1000000 / duration;

  Xiaogyan.speaker.setTone(freq);
  delay(duration);
  Xiaogyan.speaker.setTone(0);
}
~~~

コンパイルエラーは以下の通りです。
undefined reference to `XiaogyanClass::speaker'となってしまいます。
~~~
/Users/mutoh/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: sketch/main.cpp.o: in function `playToneTask(void*)':
/Volumes/mutoh/work/github/XIAOGYANbitLess/src/main.cpp:580: undefined reference to `XiaogyanClass::speaker'
/Users/mutoh/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /Volumes/mutoh/work/github/XIAOGYANbitLess/src/main.cpp:580: undefined reference to `XiaogyanClass::speaker'
/Users/mutoh/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /Volumes/mutoh/work/github/XIAOGYANbitLess/src/main.cpp:581: undefined reference to `XiaogyanClass::speaker'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "BLEDevice.h"
  Used: /Users/mutoh/Library/Arduino15/packages/esp32/hardware/esp32/2.0.8/libraries/BLE
  Not used: /Users/mutoh/Documents/Arduino/libraries/Seeed_Arduino_rpcBLE
exit status 1

Compilation error: exit status 1
~~~

同じコードをPlatformIOでコンパイルすると、無事に終了します。

理由がわかる方、おられますでしょうか?

matsujirushi, 2023-04-27:
**Xiaogyan.cpp**をコンパイル時に`SPEAKER`がdefineされていないのが原因と思います。

https://github.com/algyan/xiaogyan_arduino/blob/55b46463c1de0b06da0e00ceb5f67c5adc5e81f4/src/Xiaogyan.cpp#L20-L24

Arduinoスケッチだけで対処は無理じゃないかと思います。
以下、案です。

1. ライブラリの**Xiaogyan.hpp**に下記コードを追加する。  
```cpp
#if !defined(SPEAKER)
#define SPEAKER 1
#endif
```
2. プラットフォームの**platform.txt**に`-D SPEAKER=1`を追加する。

610t, 2023-04-28:
2.の方法だと、グローバルな設定ファイルに`-D SPEAKER=1`を追加することになるんですよね?
リポジトリ内に置いたファイルを読み込んでくれるのなら、それでも良いかと思うのですが…

1.の方法で動くことは確認しました。
特に副作用などの問題が問題がないようでしたら、1の方向で対応してもらえないでしょうか?

matsujirushi, 2023-04-28:
Arduino IDEでエラーが発生する件は #47 で対応します。

closed 2023-04-28 by matsujirushi

matsujirushi, 2023-04-28:
@610t 
#47 で対応完了しました。