algyan/XIAOGYAN #57: 組立てマニュアル　JP1のはんだ付け順序

mnltake, 2023-06-06:
組立てマニュアル
2.3. 基板裏面のショートジャンパへのはんだ付け
https://github.com/algyan/XIAOGYAN/blob/main/manuals/assembly/README.md#23-%E5%9F%BA%E6%9D%BF%E8%A3%8F%E9%9D%A2%E3%81%AE%E3%82%B7%E3%83%A7%E3%83%BC%E3%83%88%E3%82%B8%E3%83%A3%E3%83%B3%E3%83%91%E3%81%B8%E3%81%AE%E3%81%AF%E3%82%93%E3%81%A0%E4%BB%98%E3%81%91
先に裏面にはんだ付けすると表面に部品を取り付けるときに基板が不安定になるので最後にしたほうがよさそう（個人の感想です）

tatsu1225, 2023-06-11:
すみません、コメント遅くなりました．組立マニュアルを作成した者です．
確かに，ハンダを多めに盛ってしまうとガタつく可能性がありますね．
最後に裏面のハンダですと，実装したマイコンやエンコーダなどの背の高い部品が邪魔になりそうです．

変更するとしたら, 2.3と2.4の入替でしょうか．表面の表面実装が終わった後に，裏面のショートジャンパのような流れ．
> 　2.3 基板裏面のショートジャンパへのはんだ付け
> 　2.4 表面実装品のはんだ付け
> 　2.5 挿入実装品のはんだ付け

mnltake, 2023-06-11:
ありがとうございます。そうですね表面に高い部品が付く前のほうがいいですね

ちょうどJP1が基板中央にあるのでピンセットで表面に置くような部品のときにガタつくのが気になったもので

closed 2023-06-12 by mnltake