algyan/XIAOGYAN #61: 組み立てマニュアル Groveコネクタのはんだ付け

matsujirushi, 2023-06-11:
### Discussed in https://github.com/algyan/XIAOGYAN/discussions/58

<div type='discussions-op-text'>

<sup>Originally posted by **shoko93** June 11, 2023</sup>
Issueに書くか迷いましたがひとまずこちらに投稿しておきます。
組み立てマニュアルの2.5.4. Groveコネクタのはんだ付けについて。
秋月で購入したＬ型のGroveコネクタの場合そのままだと↓の写真の通りにはならないので、少し迷いました。
[image]
（私の場合はよくわからなかったので、L字の足を無理やりまっすぐにしました。。）</div>

tatsu1225, 2023-06-11:
これは私のやらかしですね．誤ってI字のGroveコネクタで実装してしまったようです．
全部の写真の差し替えは難しいですが，指摘頂いた箇所については撮影して修正します

tatsu1225, 2023-06-11:
こんな感じでどうでしょう？
[image]

closed 2023-06-12 by matsujirushi