algyan/XIAOGYAN #68: 2023/8/5現在 秋月電子でNJM2113Dが売り切れ中です

yak1ex, 2023-08-05:
その結果、「秋月電子通商 部品を買い物かごに入れる」のリンクを踏むと以下の表示が出ます。

[image]

別店舗での購入以外の手段としては、
* 回路図、部品表、基板を参照したところNJM2113M(I-10981)での代替を想定されているということでよろしいでしょうか？(※こちらも通販在庫数：12ですが)
* またこちらは自己責任かもしれませんがピン互換と思われるMC34119(I-17430)での対応もありでしょうか？ ref. http://tmj114.web.fc2.com/amp/miniamp.html

matsujirushi, 2023-08-05:
情報ありがとうございます！

* 代わりに、`NJM2113M(I-10981)`でもOKです。
* `MC34119(I-17430)`は未確認です。

（だれか`MC34119(I-17430)`で試したら、結果を教えてください...）

yamati-kz, 2023-08-06:
違和感ないです・MC34119(I-17430)を試しましたが聴いた感じでは...
本来の部品リスト上、NJM2113D(I-11192がが在庫不足ということで代替を試してみました。石が代わってもそれなりに動くのかなと思いつつ。オシロで違いを明示できればいいのですが、そんな技術者なく...
HardwareTest.binで試しています。平ピンICソケットなので載せ替えはすぐにできます。

yamati-kz, 2023-08-09:
・ご参考になるかわかりませんがオシロで波形を取ってみました。
 　　　部品リスト品 : [NJM2113D(I-11192)](https://akizukidenshi.com/catalog/g/gI-11192/)
　　　代替候補品 : [MC34119(I-17430)](https://akizukidenshi.com/catalog/g/gI-17430/)

・IC PinNo.8(VOUT2)とGND間の波形です。ボリュームはMAX(右側に回し切った状態)。

・NJM2113DでボタンＡを押したとき。
[image]

・NJM2113DでボタンBを押したとき。
[image]

・MC34119でボタンＡを押したとき。
[image]

・MC34119でボタンＢを押したとき。
[image]

matsujirushi, 2023-08-09:
@yamati-kz 
すばらしい！確認ありがとうございます。

もし可能でしたら、[PlayWave](https://github.com/algyan/xiaogyan_examples/tree/main/platformio/PlayWave)も確認していただけませんか？（聴いた感じの比較で結構ですので）

yamati-kz, 2023-08-09:
@matsujirushi 
聴いた感じ区別できません、私のような素人の耳では聞き分けできませんでした。
[PlayWave](https://github.com/algyan/xiaogyan_examples/tree/main/platformio/PlayWave)でのホワイトノイズの感じとか、とても可愛いお子ちゃまの声の感じとか、音量も違わないといいますか...

yak1ex, 2023-08-12:
在庫復活しておりました。
MC34119も買っていましたので安心して使用できます。ありがとうございます。

closed 2023-08-12 by matsujirushi